robotpy/robotpy-ctre #6: CANTalon ControlMode.Follower issue in RobotPy sim

NoMod-Programming, 2017-03-22:
Hello. While working on some robot code in RobotPy with the robotpy-ctre library, I encountered a strange bug that seems to work on the c++ and java counterparts, but not here. The code that triggers the error, at the bare minimum, is running `setControlMode(CANTalon.ControlMode.Follower)` on a CANTalon motor, and then setting it to follow another talon. The code runs through fine until it gets to initializing the simulator window. When doing so, the following is shown in the console (Using eclipse):

    18:31:33:681 INFO    : wpilib              : WPILib version 2017.1.1
    18:31:33:681 INFO    : wpilib              : HAL base version 2017.1.1; sim platform version 2017.1.1
    18:31:33:681 INFO    : wpilib              : robotpy-ctre version 2017.0.2
    18:31:33:697 WARNING : pyfrc.sim           : sim/config.json not found, using default simulation parameters
    18:31:33:701 WARNING : pyfrc.physics       : Cannot enable physics support, C:\Users\empedemonte20\workspace\Potential\src\physics.py not found
    18:31:33:705 INFO    : nt                  : NetworkTables 2017.0.6 initialized in server mode
    18:31:33:705 ERROR   : wpilib.ds           : Joystick Button 1 on port 1 not available, check if controller is plugged in
    18:31:34:331 INFO    : pyfrc.sim.ui        : pygame was detected, real joystick support loaded!
    Traceback (most recent call last):
      File "C:\Users\empedemonte20\workspace\Potential\src\robot.py", line 52, in <module>
        wpilib.run(MyRobot)
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\wpilib\_impl\main.py", line 121, in run
        retval = options.cmdobj.run(options, robot_class, **kwargs)
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\pyfrc\mains\cli_sim.py", line 107, in run
        ui = sim.SimUI(sim_manager, fake_time, config_obj)
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\pyfrc\sim\ui.py", line 95, in __init__
        self.timer_fired()
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\pyfrc\sim\ui.py", line 416, in timer_fired
        self.update_widgets()
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\pyfrc\sim\ui.py", line 492, in update_widgets
        extension.update_tk_widgets(self)
      File "C:\Users\empedemonte20\AppData\Roaming\Python\Python35\site-packages\ctre\_impl\sim_ui.py", line 51, in update_tk_widgets
        motor.set_value(self.data[data['value']][0].get_value())
    AttributeError: 'CtreUI' object has no attribute 'data'

virtuald, 2017-03-22:
that just means the simulator doesn't support it -- however, it should work on a real robot. It should be pretty straightforward to extend the simulation support to not crash however.

virtuald, 2017-03-22:
Fixed in robotpy-ctre 2017.0.3.

commit e38f496a (2017-03-22): Fix simulation issue in follower mode

- Fixes #6
closed 2017-03-22 by virtuald