robotpy/robotpy-ctre #8: Add current limiting functions

nickschatz, 2017-08-17:
`setCurrentLimit` function, giving the current limit in Amperes.

`enableCurrentLimit` function, enabling/disabling the current limiting function.

9.3 in the software reference manual.

Chris2fourlaw, 2017-09-02:
I second this, as this is a very important feature to protect our hardware.

virtuald, 2017-09-02:
I'm hoping CTRE will rearchitect their plugins such that we can support all APIs in the future. Right now it's too easy to miss things.

nickschatz, 2017-09-02:
Here's my go at it: https://github.com/nickschatz/robotpy-ctre  
The handles are all there, it just needs to be attached in Python.
I don't have ready access to a rio/talon anytime soon, so it would be great if you can test this for me.

[Compiled wheel](https://www.dropbox.com/s/tgxhqbms9s1wneh/robotpy_ctre-2017.0.3.post0.dev2-cp36-cp36m-linux_armv7l.whl?dl=1)

Chris2fourlaw, 2017-09-02:
Got it, thanks! Will be working on it today

Chris2fourlaw, 2017-09-02:
My tests passed, I'll be deploying to a rio later and testing.

Chris2fourlaw, 2017-09-02:
I can build successfully on my computer, but I'm having trouble getting it working on the rio. @virtuald How do you add support to the roborio ctre?

Chris2fourlaw, 2017-09-02:
@nickschatz where did you find the hal_data param for the current limit? Also, where do I use the whl file? Is that the file that defines the current stuff

nickschatz, 2017-09-03:
@Chris2fourlaw I can't link because I'm on mobile, but it's all in the cantalon_roborio.cpp file. The wheel file is the compiled code from my repo. You can use pip to install that on your roborio and get my current limiting code. You can't just toss this code in the rio, it needs to be compiled. I can share my setup for that if you want, because it took me all of forever to figure it out.

Chris2fourlaw, 2017-09-03:
@nickschatz Oh yeah, I see it now. I'll have to try again later when I have access to the rio. I'd also love to see your setup.

nickschatz, 2017-09-03:
https://gist.github.com/nickschatz/0c9872be5195e8306383d940ff5094b2

Here's what I needed to do to compile robotpy-ctre. Like I said, most things are implemented in C++ and just need the python wrapper functions and sim support added.

virtuald, 2017-09-03:
I've sent an inquiry to CTRE about the status of their updated API, which should make it significantly easier to add support for the entire available API. I'll wait until I hear from them about that -- though if @nickschatz wants to PR his changes, I'll merge it.

nickschatz, 2017-09-11:
@Chris2fourlaw Have you tested my changes? I'll probably get to it tomorrow if you haven't.

Chris2fourlaw, 2017-09-12:
Oh sorry! Yeah I did and I believe it’s working. What’s the best way to confirm?

nickschatz, 2017-09-12:
For starters, not crashing is a good sign. I would probably put a low limit on it and then just stall out a CIM at low power while monitoring current.

Chris2fourlaw, 2017-09-12:
Alright, well it didn’t crash. I can test the current on Thursday, but if you can do it earlier that’s better.

PR #15 (2017-09-12): Add current limiting functions
Resolves #8
closed 2017-09-12 by virtuald