robotpy/robotpy-ctre #12: Rewrite to support new CTRE drivers

virtuald, 2017-09-03:
CTRE told me during the 2017 season that they were exploring an approach that would make it easier to create a third-party wrapper around their drivers. I anticipate that a small rewrite of this package will be necessary -- but it should be fairly straightforward to do.

I've reached out to them about the status of the new drivers.

virtuald, 2017-09-12:
I've confirmed with them that an easier to wrap driver will available this season -- perhaps as early as next week. They said announcements will be made via Twitter/Facebook.

virtuald, 2017-09-17:
Looks like their framework is out now, needs someone to go look at it and see what needs to be done.

http://www.ctr-electronics.com/control-system/hro.html#product_tabs_technical_resources

virtuald, 2017-09-17:
I took a brief look, and it looks like not much has changed, so I'll talk to them to see if we can fix it so it isn't as annoying.

Twinters007, 2017-09-18:
Did you mean to link to the HERO board page?

virtuald, 2017-09-18:
Yes, if you click on the toolsuite installer link it takes you to the HERO page.

virtuald, 2017-12-04:
The latest beta installer has the Java sources included, and it looks like it's easier to wrap now. Anyone want to take a crack at it?

Twinters007, 2017-12-04:
I'll take a look, but can't for sure say I Can dedicate enough time to do it before someone else does

virtuald, 2017-12-07:
I think the best approach would be trying to autogenerate the bindings from the available header files... as I think they've removed most of the magic from the java code now.

ariovistus, 2018-01-03:
I can take a crack at this if you don't have time. I assume pygccxml on the cpp include files and something something ctypes

virtuald, 2018-01-03:
I've already done a large chunk of it, my goal is to entirely autogenerate the motor controller bindings from the C stuff. Check out the autogen branch.

Developing https://github.com/virtuald/header2whatever for codegen, it's pretty neat.

ariovistus, 2018-01-03:
shall I work on the library part?

virtuald, 2018-01-03:
The library part? Which part is that? I'm not planning to implement the phoenix framework stuff.

virtuald, 2018-01-03:
I'll be on gitter for about an hour if you want to yap about this.

PR #19 (2018-01-13): Initial hardware-only 2018 support
I don't have hardware, so I can't confirm that this actually works. If someone can confirm that it works on hardware, I'll push an initial release so that people can start using it.

* Fixes #1
* Fixes #4 (well... once CTRE re-releases it?)
* Fixes #9
* Fixes #10
* Fixes #11
* Fixes #12
* Fixes #13
* Fixes #17

Here's an initial compiled wheel that I've installed on a RoboRIO and it doesn't seem to crash -- download the file to the RoboRIO, and execute 'pip3 install filename.whl'. The final release will be in opkg form.

https://www.tortall.net/~robotpy/feeds [... cut]
closed 2018-01-13 by virtuald