robotpy/robotpy-ctre #22: BaseMotorController.getMotorOutputVoltage attempts to multiply 2 tuples and throws an error

vanjac, 2018-01-13:
I'm testing with 2018.0.0.

Part of the traceback:

```
File "/home/lvuser/py/motortest.py", line 112, in updateTalonLog 
     self.outputVoltageLog.update(talon.getMotorOutputVoltage()) 
   File "/usr/local/lib/python3.6/site-packages/ctre/basemotorcontroller.py", line 154, in getMotorOutputVoltage 
     return self.getBusVoltage() * self.getMotorOutputPercent() 
 TypeError: can't multiply sequence by non-int of type 'tuple' 
```

vanjac, 2018-01-13:
A lot of talon functions (getOutputCurrent, getSelectedSensorPosition, getSelectedSensorVelocity) are returning tuples like: `(ErrorCode.RxTimeout, 0)`

vanjac, 2018-01-13:
And all the functions that return an error code like that print this message to driver station:
```
ERROR  -3  CTR: CAN frame not received/too-stale.  Talon SRX 0 GetOutputCurrent
```

virtuald, 2018-01-14:
Strictly speaking, it is documented: http://robotpy.readthedocs.io/projects/ctre/en/stable/api.html#ctre.talonsrx.TalonSRX.getMotorOutputPercent

... but it could be better.

virtuald, 2018-01-14:
Also, I've reached out to CTRE to see how we should deal with this.... currently, I always return the error code so that the user can use it. However, we could just throw an exception if there's an error code (assuming that a returned error is always a bad thing). Or I suppose we could ignore it, since there is a getLastError function.

vanjac, 2018-01-14:
Other versions of the library don't return the error code:

- [C++](http://www.ctr-electronics.com/downloads/api/cpp/html/classctre_1_1phoenix_1_1motorcontrol_1_1can_1_1_base_motor_controller.html#a8d9e038c79a96df3ca7ced4d5a6e468b)
- [Java](http://www.ctr-electronics.com/downloads/api/java/html/com/ctre/phoenix/motorcontrol/can/BaseMotorController.html#getMotorOutputPercent--)

And it seems like parts of robotpy-ctre itself are not expecting the tuple, like getMotorOutputVoltage.

auscompgeek, 2018-01-14:
I guess we should handle the error code in these getter functions in our wrapper then, since the CTRE framework throws them away there. (Maybe add a thing in the autogen YAML config saying "check error code" or something.)

virtuald, 2018-01-14:
From CTRE:

> For the most part, it's ignored.  We do some additional checking in Java and LabVIEW so we can print a stack trace with our DS message, but that's about it.  We don't generate exceptions with them - we just pass them on to the user (provided there's nothing else that function needs to return - otherwise users have getLastError).  Their purpose is basically to (1) Generate a DS error so the user knows something Bad happened, and (2) give users a way in the API to handle it in code if they want to.

virtuald, 2018-01-16:
I've got some fixes done for this, I'll hopefully push them tonight, along with some initial simulation support.

commit 4ef6b6ba (2018-01-17): Update code generators

- Match CTRE Java API by suppressing return code when other output params are present (Fixes #22)
- Add typechecking on simulation return values
commit a416e547 (2018-01-18): Update code generators

- Match CTRE Java API by suppressing return code when other output params are present (Fixes #22)
- Add typechecking on simulation return values
closed 2018-01-18 by virtuald