robotpy/robotpy-ctre #25: Release v4_Legacy compatible package?

auscompgeek, 2018-01-15:
I have something over at https://github.com/auscompgeek/robotpy-ctre/tree/legacy which will probably work? I'm sure some teams would appreciate this.

auscompgeek, 2018-01-18:
So I think I have something that builds now. Hopefully it'll build and work on a roboRIO?

(Also waiting on robotpy/robotpy-wpilib#430 to be merged. Looks like CTRE didn't provide source for the Java bit of v4_Legacy, and I'm too lazy to decompile it right now, so I'm not touching the actual source.)

auscompgeek, 2018-02-04:
Oh yeah. I got this compiled for the roboRIO last week. If anyone would like to test this out (and trust me to not cause your roboRIO to eat your cat), be my guest (and please tell us if you actually want to use this!): https://vovo.id.au/tmp/robotpy_ctre_legacy-2017.0.3.post0.dev15-cp36-cp36m-linux_armv7l.whl

(disclaimer: I don't actually know whether this works)

virtuald, 2018-04-24:
I think since the season is almost over, this is OBE.

closed 2018-04-24 by virtuald