robotpy/robotpy-ctre #33: setPulseWidthPosition

adam-mcdaniel, 2018-01-23:
I cant seem to use the setPulseWidthPosition method on the WPI_TalonSRX class. I'm trying to use it to zero the position, but when I use it, it doesnt change a thing. Is there another function I should be using to zero the Pulse Width Position, or is there a problem?

adam-mcdaniel, 2018-01-23:
I just changed the program to use setQuadraturePosition and getQuadraturePosition and the program works as intended. It seems the setPulseWidthPosition function is broken.

virtuald, 2018-01-24:
Assuming that you're talking about real device usage and not simulation, that function is implemented directly in the CTRE library, so if it doesn't work it's a bug in their implementation. I would file a bug at https://github.com/CrossTheRoadElec/Phoenix-frc-lib/issues and reference this issue.

virtuald, 2018-01-24:
Actually, one thing you might try -- does setPulseWidthPosition return an error code that isn't `ErrorCode.OK`? That could be instructive.

closed 2018-02-09 by virtuald