robotpy/robotpy-ctre #45: 1ms Notifier significantly slows down tests

auscompgeek, 2018-02-12:
Running pyfrc tests now end up taking about a minute or so, when they previously finished under 20 seconds.

I think having the simulated PID loop run every 10ms instead (similarly to the simulator) would be more sensible.

virtuald, 2018-02-12:
@ariovistus would that negatively affect the math?

auscompgeek, 2018-02-13:
I imagine we would have to compensate for the loop running at a 10th of what it is currently at, but otherwise I imagine it'd positively affect the maths, as the physics.py would be running at the same rate (and so would be able to react to the PID loop in a timely manner).

ariovistus, 2018-02-13:
a 10ms pid takes different parameters as a 1ms loop for equivalent functionality. I don't know how to predict what the difference would be, or even if it's possible.

auscompgeek, 2018-02-13:
Given that the simulator will only update every ~10ms, a 1ms PID loop won't respond correctly in the simulator anyway.

auscompgeek, 2018-04-08:
Update: with the game data cycler and all (9) our autonomous modes, my team's autonomous tests (and _only_ the autonomous tests) now take over 12 minutes in total to run on my laptop. (By my calculations, this means our auto tests take longer than real time!)

The usual pyfrc tests, minus the auto tests (so test_disabled, test_operator_control, test_practice, test_fuzz), take about a minute on my laptop.

ariovistus, 2018-04-08:
does 

```
ctre.WPI_TalonSRX.Notifier = None
```

help any?

vanjac, 2019-02-10:
We had to disable the notifier because it was causing the simulator to run very slowly and sometimes freeze.

auscompgeek, 2019-02-10:
FYI for anyone else who needs to disable the Talon PID simulation, set the `_use_notifier` attribute on the TalonSRX object to `False`.

auscompgeek, 2019-02-10:
Thinking a bit about how we could go about fixing this, it'd be good to have a way for plugins to hook into the pyfrc physics loop for predictability.

auscompgeek, 2020-01-27:
OBE with the scrapping of our simulation.

closed 2020-01-27 by auscompgeek