robotpy/robotpy-ctre #61: configRemoteFeedbackFilter throws NotImplementedError. Is there a quick fix or workaround for this?

stephonlawrence, 2018-04-17:
I'm trying to do Motion Profiling with 4 Talon_SRX's, 2 quadrature encoders and a pigeon_imu.
encoder 1 is plugged into talon_srx_1
encoder 2 is plugged into talon_srx_2
pigeon_imu is plugged into talon_srx_3
and talons 2, 3, 4 are set to follow talon one.

I'm trying to use only talon_1 for it's PID loop for encoders and aux PID loop for the pigeon.

Pretty much following this example from the java code: https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/blob/master/Java/RemoteClosedLoop/src/org/usfirst/frc/team4130/robot/Robot.java

But it seems like this function is not fully implemented yet in the python api. Is there a quick fix or workaround for this?

virtuald, 2018-04-17:
The entire API is supported on the RoboRIO... so it should work on your robot.

However, due to their complexity not all functions are supported in simulation yet. Just deploy with `--skip-tests` or add something to skip your motion profiling when simulating.

stephonlawrence, 2018-04-17:
So I tried deploying code to the robot but I get a `notImplemented` error when using the `configRemoteFeedbackFilter` method.

Here's a snippet showing where I'm doing this.

Here are the constants:
```
PID_PRIMARY = 0
REMOTE_0 = 0
REMOTE_1 = 1
PID_TURN = 1
TimeOut_ms = 0
```

```
left_front = WPI_TalonSRX(5)
self.left_encoder_motor = left_rear = WPI_TalonSRX(1)
self.right_encoder_motor = right_front = WPI_TalonSRX(4)
self.pigeon_talon = right_rear = WPI_TalonSRX(6)
self.pigeon = PigeonIMU(self.pigeon_talon)

# setup left encoder
self.left_encoder_motor.configSelectedFeedbackSensor(
    FeedbackDevice.QuadEncoder,
    DriverComponent.PID_PRIMARY,
    DriverComponent.TimeOut_ms
)
# connect left encoder talon to right talon srx
self.right_encoder_motor.configRemoteFeedbackFilter(
    self.left_encoder_motor.getDeviceID(),
    RemoteSensorSource.TalonSRX_SelectedSensor,
    DriverComponent.REMOTE_0,
    DriverComponent.TimeOut_ms
)
# connect piegon talon to right talon srx
self.right_encoder_motor.configRemoteFeedbackFilter(
    self.pigeon_talon.getDeviceID(),
    RemoteSensorSource.GadgeteerPigeon_Yaw,
    DriverComponent.TimeOut_ms
)
```

Because RemoteSensorSource also wasn't implements I just created a quick class with this info. 
```
class RemoteSensorSource(object):
    Off = int(0)
    TalonSRX_SelectedSensor = int(1)
    Pigeon_Yaw = int(2)
    Pigeon_Pitch = int(3)
    Pigeon_Roll = int(4)
    CANifier_Quadrature = int(5)
    CANifier_PWMInput0 = int(6)
    CANifier_PWMInput1 = int(7)
    CANifier_PWMInput2 = int(8)
    CANifier_PWMInput3 = int(9)
    GadgeteerPigeon_Yaw = int(10)
    GadgeteerPigeon_Pitch = int(11)
    GadgeteerPigeon_Roll = int(12)
```

I already use the  `--skip-tests` option when deploying. I am not running any simulation at the moment.

auscompgeek, 2018-04-18:
Yes, configRemoteFeedbackFilter is not implemented _for simulation_ yet. It'll work on your roboRIO fine.

We actually do have the RemoteSensorSource enum. However, due to an oversight we forgot to expose it. (Feel free to send a PR to fix this, otherwise it'll be fixed eventually.)

virtuald, 2018-04-18:
@stephonlawrence how did you install robotpy-ctre on the RoboRIO? Did you install it via pip or via the robotpy-installer 'install-opkg' functionality.

virtuald, 2018-04-18:
With robotpy-ctre 2018.3.0 installed, I was able to run your code without problems (except that your second call to `configRemoteFeedbackFilter` is missing the REMOTE_ argument).

I believe you have the simulation version of robotpy-ctre installed on your roborio. We tried to make it so you couldn't do that... what method did you use to install it so we can fix it?

See http://robotpy.readthedocs.io/en/stable/install/ctre.html#python-package for installation procedure. You will need to first uninstall the version of robotpy-ctre that you installed on your roborio via `pip3 uninstall robotpy-ctre`.