robotpy/robotpy-ctre #115: Implement configFactoryDefault

vanjac, 2019-03-16:
This is a useful function for resetting all parameters of a talon to a known state. It crashes in the simulator right now. At least for us, it doesn't need to do anything in the simulator, just not crash.

virtuald, 2019-03-18:
A workaround is:

```
try:
    motor.configFactoryDefault()
except NotImplementedError:
    pass
```

I realize it's a bit verbose (and perhaps the exception error message should display this?), but this particular function I'm reluctant to make it 'pretend to work' without actually making it do what it should do (as this shouldn't be too difficult to implement). Soon perhaps?

auscompgeek, 2019-03-18:
I might get around to doing this in the next couple of days.

auscompgeek, 2019-10-03:
Evidently I never got around to actually doing this.

Probably the cleanest way of doing this is to do what I did for robotpy-rev and use the `configFactoryDefault` method to also initialise the configs in the constructor.

auscompgeek, 2020-01-27:
I presume this is OBE and CTRE's simulation libraries do have a sane implementation of configFactoryDefault.

closed 2020-01-27 by auscompgeek