robotpy/robotpy-ctre #118: WPI_TalonFX support

adam-mcdaniel, 2020-01-06:
In the new CTRE API, Falcon-500s have been added to the mix. Our team needs to be able to prototype with these motors soon, and right now the only option is another programming language.

Is the Falcon 500 implementation already underway?

You can find these motors [here](http://www.ctr-electronics.com/talon-fx.html).

virtuald, 2020-01-06:
We're switching to autogenerated C++ wrappers this year, so the implementation for the 2020 CTRE libraries should come shortly after WPILib is available (as the CTRE headers are fairly simple in comparison to some of the WPILib code).

I refer you to https://www.chiefdelphi.com/t/robotpy-2020-availability/369479 for the current status of RobotPy 2020.

virtuald, 2020-01-23:
Fixed in 2020.0.0, Windows sim binaries should be on pypi in a few minutes, expect a roborio release sometime tomorrow.

closed 2020-01-23 by virtuald