robotpy/robotpy-ctre #119: 404 Not found newest version

ryowl, 2020-01-27:
For some reason in the last week we've had issues reinstalling python-ctre on our computer. It seems to install onto the robot fine but we can't deploy.

When I try to use pip to install it, I get this error.
[image]

It seems it can't find that specific zip file on the CTRE website, but is that something we can fix on my end?

auscompgeek, 2020-01-27:
CTRE do not support simulation on x86, and have no plans to do so. Please use an x86-64 installation of Python, or upgrade your computer if that is not possible (or complain to CTRE).

ryowl, 2020-01-27:
Okay I will try that. Did something change recently that would have caused this issue because about a week or two ago the code worked fine and our CTRE Talons worked great, but because of some issues we had to uninstall and reinstall python-ctre and this occured.

virtuald, 2020-01-27:
You were most likely using the 2019 code previously.

closed 2020-01-27 by virtuald