robotpy/robotpy-ctre #124: error: HTTP Error 404: Not Found

porgull, 2020-02-14:
```
$ pip3 install -U robotpy-ctre
Collecting robotpy-ctre
  Downloading https://files.pythonhosted.org/packages/03/fb/28651f0540edeed6eafcd6a7589dbc2e94dd99142cbc112041b1df8f47dc/robotpy-ctre-2020.2.1.tar.gz
  Installing build dependencies: started
  Installing build dependencies: still running...
  Installing build dependencies: finished with status 'done'
  WARNING: Missing build requirements in pyproject.toml for robotpy-ctre from https://files.pythonhosted.org/packages/03/fb/28651f0540edeed6eafcd6a7589dbc2e94dd99142cbc112041b1df8f47dc/robotpy-ctre-2020.2.1.tar.gz#sha256=02b98d4c9843011280d022b165dd9dfb6167d786fefecbbf32da32ca6c3a1938.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Installing backend dependencies: started
  Installing backend dependencies: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Collecting wpilib<2021.0.0,>=2020.2.2.0
  Using cached https://files.pythonhosted.org/packages/ef/a1/08852fe60b2230176605a2b83c3a3d5f7cf92b51c996194fcd948eaeb8fc/wpilib-2020.2.2.10-cp37-cp37m-win32.whl
Collecting pyntcore<2021.0.0,>=2020.2.2.0
  Using cached https://files.pythonhosted.org/packages/8e/83/3c3c4744152bc780416782194c74923d2529b7875ebcd6b375a93c990468/pyntcore-2020.2.2.3-cp37-cp37m-win32.whl
Collecting robotpy-wpiutil<2021.0.0,>=2020.2.2.0
  Using cached https://files.pythonhosted.org/packages/df/67/b52502cb3fd62f316087dfadf4390122a774e0fb2e653100646fe0c1b042/robotpy_wpiutil-2020.2.2.2-cp37-cp37m-win32.whl
Collecting robotpy-hal<2021.0.0,>=2020.2.2.0
  Using cached https://files.pythonhosted.org/packages/d0/77/3cf21f6974d9452cc66da9e9b6def551b7ce8a43b6230bd1c5ebebc16dbb/robotpy_hal-2020.2.2.2-cp37-cp37m-win32.whl
Collecting pynetworktables<2021.0.0,>=2020.0.0
  Using cached https://files.pythonhosted.org/packages/ef/89/8947a4fe98a3f199271b16682834ed0338c248988e1ee0a85daa844c8ee4/pynetworktables-2020.0.2-py3-none-any.whl
Building wheels for collected packages: robotpy-ctre
  Building wheel for robotpy-ctre (PEP 517): started
  Building wheel for robotpy-ctre (PEP 517): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\connor barker\appdata\local\programs\python\python37-32\python.exe' 'c:\users\connor barker\appdata\local\programs\python\python37-32\lib\site-packages\pip\_vendor\pep517\_in_process.py' build_wheel 'C:\Users\CONNOR~1\AppData\Local\Temp\tmplvxotdps'
       cwd: C:\Users\CONNOR~1\AppData\Local\Temp\pip-install-s7n2t3f_\robotpy-ctre
  Complete output (8 lines):
  running bdist_wheel
  running build
  running build_py
  running build_gen
  running build_dl
  Downloading http://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/api-cpp/5.18.1/api-cpp-5.18.1-headers.zip
  100%Downloading http://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/api-cpp/5.18.1/api-cpp-5.18.1-windowsx86static.zip
  error: HTTP Error 404: Not Found
  ----------------------------------------
  ERROR: Failed building wheel for robotpy-ctre
  Running setup.py clean for robotpy-ctre
Failed to build robotpy-ctre
ERROR: Could not build wheels for robotpy-ctre which use PEP 517 and cannot be installed directly
```

I had an error while installing over pip. Looks like the url is malformed - this link works:

http://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/api-cpp/5.18.1/api-cpp-5.18.1-windowsx86-64static.zip

auscompgeek, 2020-02-14:
Duplicate of #119.

closed 2020-02-14 by auscompgeek

virtuald, 2020-02-14:
@porgull CTRE isn't compatible with 32-bit python. Please uninstall your 32-bit python and install a 64-bit python, and CTRE installation should work. Thanks!