robotpy/robotpy-ctre #140: Unhiding the Unmanaged class from CTRE Package Docs

lisianshehu, 2021-06-07:
Recently, my group and I were utilizing the robotpy wrapper library for the non-FRC CTRE library to control Talon SRX's and we learned that we had to use the `feedEnable` function in the `Unmanaged` class to enable the Talons. This class was not under the CTRE package API list and had to be searched for in the docs in order to be found(seemingly the only way to access the docs for this class). It would be useful(I think) for future users to be able to find this class easily by adding it to the CTRE package list (https://robotpy.readthedocs.io/projects/ctre/en/stable/ctre.html).

auscompgeek, 2021-06-07:
Hm, that's weird, Read the Docs isn't serving docs for the latest version, despite it having built docs for the latest version. I'm not too sure what's going on here, @virtuald do you have any ideas?

auscompgeek, 2021-06-07:
Super weird that it seems to have merged docs for different versions together if you search for the `Unmanaged` class.

lisianshehu, 2021-06-07:
Interesting so is the `Unmanaged` class supposed to be included in the package list without the need to be searched for?

brandons209, 2021-06-07:
Yeah I am apart of the same team with lisian, maybe this class should be pointed out in a tutorial page or a non-FRC code example should be added for users who aren't using the roborio and WPI.

auscompgeek, 2021-06-07:
We'll happily accept Python translations of any of CTRE's examples; see #53.

virtuald, 2021-06-08:
Not sure how it got broken, but I rebuilt the packages and they're correct now. Thanks for reporting the issue!

closed 2021-06-08 by virtuald