robotpy/robotpy-ctre #142: Configuring WPI_TalonFX results in segmentation fault

FRC2539, 2022-01-20:
### Problem description

First, all of our motors and encoders have been updated to the latest firmware version (22) via Phoenix Tuner.

We noticed there was a segmentation fault, and to locate the issue, we commented out the code to start with a basic robot.py file. In the robot init, we initialize a Talon. That by itself results in no segmentation fault. We added `motor.setNeutralMode(NeutralMode.Brake)`, and that was all it took to produce the segmentation fault.

Here is the segmentation fault trace:

[image]

### Operating System

RoboRIO

### Reproducible example code

```text
motor = WPI_TalonFX(0)

motor.setNeutralMode(NeutralMode.Brake)
```

virtuald, 2022-01-20:
Can reproduce in simulation also on OSX.

virtuald, 2022-01-20:
Tried it in C++, works fine, so it must be something on our end. Unfortunately, I'm away from my desktop for the night, so I won't be able to look into this until tomorrow morning. Sorry about that.

FRC2539, 2022-01-20:
No problem 👍

TheTripleV, 2022-01-20:
Seems to occur on everything inheriting from BaseTalon. BaseTalon works though.

virtuald, 2022-01-20:
It's a multple virtual inheritance thing... I've got tests duplicating the issue at https://github.com/robotpy/robotpy-build/pull/156

commit ed68397a (2022-01-22): Inform pybind11 of multiple inheritance, upgrade robotpy-build/pybind11

- Fixes #142
PR #143 (2022-01-22): Inform pybind11 of multiple inheritance, upgrade robotpy-build/pybind11
- Fixes #142
closed 2022-01-22 by virtuald