robotpy/robotpy-ctre #147: [BUG]: CANdleConfiguration has no CANdle methods

KenwoodFox, 2022-02-03:
### Problem description

It looks like the `CANdleConfiguration()` class only has methods from the parent class (https://store.ctr-electronics.com/content/api/java/html/classcom_1_1ctre_1_1phoenix_1_1_custom_param_configuration.html#a4d20c307bca4e9861b9e2a828b360a59) i think, and none of the custom values for anything else? its possible im just using the library wrong.

Heres the api for the CANdleConfiguration() https://store.ctr-electronics.com/content/api/java/html/classcom_1_1ctre_1_1phoenix_1_1led_1_1_c_a_ndle_configuration.html

Heres my example code that does not work: https://github.com/FRC-1721/pre2022season/blob/7d918611b625befc98f916542a78c354b9a31599/rio/subsystems/lighting.py#L27-L30

When uncommented heres the full error: 
```
make sim
python robot.py sim
09:30:42:312 INFO    : wpilib              : WPILib version 2022.3.1.0
09:30:42:312 INFO    : wpilib              : HAL version 2022.3.1.0
09:30:42:312 INFO    : wpilib              : Running with simulated HAL.
09:30:42:313 INFO    : wpilib              : robotpy-wpimath version 2022.3.1.0
09:30:42:313 INFO    : wpilib              : robotpy-commands-v2 version 2022.3.1.0
09:30:42:313 INFO    : wpilib              : pyntcore version 2022.3.1.0
09:30:42:313 INFO    : wpilib              : robotpy-wpiutil version 2022.3.1.0
09:30:42:313 INFO    : wpilib              : robotpy-halsim-gui version 2022.3.1.0
09:30:42:313 INFO    : wpilib              : robotpy-ctre version 2022.0.5
09:30:42:313 INFO    : wpilib              : robotpy-rev version 2022.0.1
09:30:42:313 INFO    : faulthandler        : registered SIGUSR2 for PID 401614
09:30:42:314 INFO    : halsim_gui          : WPILib HAL Simulation 2022.3.1.0
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
09:30:42:582 WARNING : pyfrc.physics       : Cannot enable physics support, /home/joe/git/pre2022season/rio/physics.py not found
Not loading CameraServerShared
09:30:42:604 ERROR   : your.robot          : Unhandled exception
Traceback (most recent call last):
  File "/home/joe/.local/share/virtualenvs/rio-t_s-Nm-J/lib/python3.10/site-packages/wpilib/_impl/start.py", line 124, in _start
    self.robot.startCompetition()
  File "/home/joe/git/pre2022season/rio/robot.py", line 28, in robotInit
    self.container = RobotContainer()
  File "/home/joe/git/pre2022season/rio/robotcontainer.py", line 37, in __init__
    self.lighting = Lighting()
  File "/home/joe/git/pre2022season/rio/subsystems/lighting.py", line 29, in __init__
    CANdleConfig.configLEDType = LEDStripType.RGB
AttributeError: 'ctre._ctre.led.CANdleConfiguration' object has no attribute 'configLEDType'

Locals at innermost frame:

{ 'CANdleConfig': <ctre._ctre.led.CANdleConfiguration object at 0x7f889a8e78b0>,
  '__class__': <class 'subsystems.lighting.Lighting'>,
  'self': <subsystems.lighting.Lighting object at 0x7f889ae473d0>}

09:30:42:605 WARNING : robotpy             : The robot program quit unexpectedly. This is usually due to a code error.
The above stacktrace can help determine where the error occurred.

[phoenix-diagnostics] Server shutdown cleanly. (dur:10|0)

[phoenix] Library shutdown cleanly

[phoenix-diagnostics] Server shutdown cleanly. (dur:10|0)

make: *** [Makefile:18: sim] Error 1
```

If you `print(CANdleConfig.toString("Prepend"))` then you see this:
```
Prepend.customParam0 = 0;
Prepend.customParam1 = 0;

09:32:06:601 INFO    : root                : <class 'ctre._ctre.led.CANdleConfiguration'>
```

### Operating System

Linux

### Installed Python Packages

```text
#
# These requirements were autogenerated by pipenv
# To regenerate from the project's Pipfile, run:
#
#    pipenv lock --requirements
#

-i https://pypi.org/simple
attrs==21.4.0; python_version >= '2.7' and python_version not in '3.0, 3.1, 3.2, 3.3, 3.4'
bcrypt==3.2.0; python_version >= '3.6'
cffi==1.15.0
click==8.0.3; python_version >= '3.6'
cryptography==36.0.1; python_version >= '3.6'
iniconfig==1.1.1
packaging==21.3; python_version >= '3.6'
paramiko==2.9.2
pint==0.18; python_version >= '3.7'
pluggy==1.0.0; python_version >= '3.6'
py==1.11.0; python_version >= '2.7' and python_version not in '3.0, 3.1, 3.2, 3.3, 3.4'
pycparser==2.21
pyfrc==2022.0.2; platform_machine != 'armv7l'
pynacl==1.5.0; python_version >= '3.6'
pynetconsole==2.0.2
pynetworktables==2021.0.0; python_version >= '3.5'
pyntcore==2022.3.1.0; python_version >= '3.6'
pyparsing==3.0.7; python_version >= '3.6'
pytest==7.0.0rc1; python_version >= '3.6'
pyyaml==6.0
robotpy-commands-v2==2022.3.1.0
robotpy-ctre==2022.0.5
robotpy-hal==2022.3.1.0; python_version >= '3.6'
robotpy-halsim-gui==2022.3.1.0; platform_machine != 'armv7l'
robotpy-installer==2022.1.0; platform_machine != 'armv7l'
robotpy-rev==2022.0.1
robotpy-wpilib-utilities==2022.0.1; python_version >= '3.6'
robotpy-wpimath==2022.3.1.0; python_version >= '3.6'
robotpy-wpiutil==2022.3.1.0; python_version >= '3.6'
robotpy[commands2,ctre,rev]==2022.0.44
six==1.16.0; python_version >= '2.7' and python_version not in '3.0, 3.1, 3.2, 3.3'
tomli==2.0.0; python_version >= '3.7'
wpilib[all]==2022.3.1.0
```

### Reproducible example code

```text
myCANdleConfig = CANdleConfiguration()
myCANdleConfig.configLEDType = LEDStripType.RGB
myCANdleConfig.brightnessScalar = 0.5
```

virtuald, 2022-02-03:
This appears to be a bug in CppHeaderParser.

closed 2022-02-03 by virtuald

KenwoodFox, 2022-02-03:
Thank you so much! :heart:

commit 763bfb2b (2022-02-03): Add CANdleConfiguration properties

- Fixes #147