robotpy/robotpy-ctre #151: [BUG]: robot code on MacOS aborts on exit

virtuald, 2022-03-25:
### Problem description

Example stack trace courtesy of auscompgeek:

```
(lldb) bt
* thread #15, stop reason = signal SIGABRT
  * frame #0: 0x00007fff2074291e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff207715bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff206c6406 libsystem_c.dylib`abort + 125
    frame #3: 0x00007fff20734ef2 libc++abi.dylib`abort_message + 241
    frame #4: 0x00007fff207265e5 libc++abi.dylib`demangling_terminate_handler() + 242
    frame #5: 0x00007fff2061f595 libobjc.A.dylib`_objc_terminate() + 104
    frame #6: 0x00007fff20734307 libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007fff20736beb libc++abi.dylib`__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
    frame #8: 0x00007fff20736bb2 libc++abi.dylib`__cxa_throw + 116
    frame #9: 0x00007fff207117bb libc++.1.dylib`std::__1::__throw_system_error(int, char const*) + 77
    frame #10: 0x00007fff207083ed libc++.1.dylib`std::__1::mutex::lock() + 29
    frame #11: 0x0000000107b3cb81 libCTRE_PhoenixCCISim.dylib`___lldb_unnamed_symbol567$$libCTRE_PhoenixCCISim.dylib + 33
    frame #12: 0x0000000107b2613f libCTRE_PhoenixCCISim.dylib`___lldb_unnamed_symbol325$$libCTRE_PhoenixCCISim.dylib + 31
    frame #13: 0x0000000107b25d6e libCTRE_PhoenixCCISim.dylib`___lldb_unnamed_symbol324$$libCTRE_PhoenixCCISim.dylib + 174
    frame #14: 0x0000000107b2808c libCTRE_PhoenixCCISim.dylib`___lldb_unnamed_symbol352$$libCTRE_PhoenixCCISim.dylib + 44
    frame #15: 0x00007fff207718fc libsystem_pthread.dylib`_pthread_start + 224
    frame #16: 0x00007fff2076d443 libsystem_pthread.dylib`thread_start + 15
```

I reached out to CTRE awhile ago about this.

### Operating System

MacOS

### Reproducible example code

```text
#include <ctre/phoenix/motorcontrol/can/WPI_TalonFX.h>
#include <ctre/phoenix/cci/Unmanaged_CCI.h>
#include <frc/TimedRobot.h>

int main() {
  ctre::phoenix::motorcontrol::can::WPI_TalonFX motor(0);
  auto& collection = motor.GetSensorCollection();
  collection.SetIntegratedSensorPosition(10);
  printf("Get %f\n", collection.GetIntegratedSensorPosition());
  return 1;
}
```

auscompgeek, 2022-03-25:
Oh, should probably also leave the exception message here:
```
libc++abi: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument

virtuald, 2022-05-04:
Sounds like an upcoming release of CTRE will fix this issue, yay.

auscompgeek, 2023-02-08:
I haven't seen this happen this year so I think we're good.

auscompgeek, 2024-01-17:
Reopening since we're getting user reports, and observed this when cutting the beta releases too.

cyocom, 2024-01-18:
Was able to reproduce this in a Java project by quickly stopped/reopening the sim.
I'm in contact with CTRE and working to resolve this with them.

Stack Trace:
```
Thread 40 Crashed:
0   libsystem_kernel.dylib                   0x18399e0dc __pthread_kill + 8
1   libsystem_pthread.dylib                  0x1839d5cc0 pthread_kill + 288
2   libsystem_c.dylib                        0x1838e1a40 abort + 180
3   libc++abi.dylib                          0x18398d070 abort_message + 132
4   libc++abi.dylib                          0x18397d110 demangling_terminate_handler() + 320
5   libobjc.A.dylib                          0x18362399c _objc_terminate() + 160
6   libc++abi.dylib                          0x18398c434 std::__terminate(void (*)()) + 16
7   libc++abi.dylib                          0x18398f520 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib                          0x18398f464 __cxa_throw + 308
9   libc++.1.dylib                           0x18390b6b8 std::__1::__throw_system_error(int, char const*) + 100
10  libc++.1.dylib                           0x183900318 std::__1::mutex::lock() + 40
11  libCTRE_PhoenixTools_Sim.dylib           0x15f813d3c 0x15f78c000 + 556348
12  libCTRE_PhoenixTools_Sim.dylib           0x15f930f5c 0x15f78c000 + 1724252
13  libCTRE_PhoenixTools_Sim.dylib           0x15f934608 0x15f78c000 + 1738248
14  libsystem_pthread.dylib                  0x1839d6034 _pthread_start + 136
15  libsystem_pthread.dylib                  0x1839d0e3c thread_start + 8
```

JoshuaBrest, 2024-07-10:
@auscompgeek Will this be fixed?

auscompgeek, 2024-07-10:
@JoshuaBrest I don't know why you're asking me. I don't work at CTRE.